nusrobomaster/PX4-Autopilot #10: DBUS RC Input Support

chengguizi, 2021-01-03:
On Dev A, DBus utilise USART1_RX (PB7), and has an inverting logic (when input is high, the RX pin is low)
[image]

Similarly, for Dev C, the port is UART3_RX (PC11)
[image]

chengguizi, 2021-01-03:
Reference code 
https://github.com/RoboMaster/DevelopmentBoard-Examples/blob/master/RemoteControl/Src/main.c

https://github.com/garfield38/DJI_DBUS/blob/master/DJI_DBUS.cpp
https://github.com/RoboMaster-Club/arduino_DBUS_decoder

Blog:
https://www.codetd.com/en/article/7453572

chengguizi, 2021-01-03:
https://www.youtube.com/watch?v=IqLUHj7nJhI&ab_channel=Mad%27sTech
https://github.com/uzh-rpg/rpg_quadrotor_control/wiki/SBUS-Protocol

Good materials to learn

chengguizi, 2021-01-03:
The code base within PX4 for SBUS, may just need to take care of the inverting logic `rc_io_invert`

https://github.com/nusrobomaster/PX4-Autopilot/blob/rm2021/src/drivers/rc_input/RCInput.cpp

```cpp
/************************************************************************************
 * Name: board_rc_invert_input
 *
 * Description:
 *   All boards may optionally define RC_INVERT_INPUT(bool invert) that is
 *   used to invert the RC_SERIAL_PORT RC port (e.g. to toggle an external XOR via
 *   GPIO).
 *
 * Input Parameters:
 *   invert_on - A positive logic value, that when true (on) will set the HW in
 *               inverted NRZ mode where a MARK will be 0 and SPACE will be a 1.
 *
 * Returned Value:
 *   true the UART inversion got set.
 *
 ************************************************************************************/
```

chengguizi, 2021-01-04:
`rc_update` module
subscribe to
- input_rc
- parameter_update
- rc_parameter_map
publish to
- rc_channels
- actuator_controls_3
- manual_control_setpoint

chengguizi, 2021-01-04:
The publisher of the `rc_input` message lies within `RCInput.hpp`
`uORB::PublicationMulti<input_rc_s>	_to_input_rc{ORB_ID(input_rc)};`

chengguizi, 2021-01-04:
For figuring out which ttySx port the RC is one, probably you want to enable the following option:

```
config STM32_SERIAL_DISABLE_REORDERING
	bool "Disable reordering of ttySx devices."
	default n
	---help---
		NuttX per default reorders the serial ports (/dev/ttySx) so that the
		console is always on /dev/ttyS0. If more than one UART is in use this
		can, however, have the side-effect that all port mappings
		(hardware USART1 -> /dev/ttyS0) change if the console is moved to another
		UART. This is in particular relevant if a project uses the USB console
		in some boards and a serial console in other boards, but does not
		want the side effect of having all serial port names change when just
		the console is moved from serial to USB.
```

chengguizi, 2021-01-04:
This seems to be the parsing code we had last year

https://github.com/nusrobomaster/robot_firmware/blob/f01669813e0cd866af75f9e16014deb0f9b56f38/standard_robot/Core/Tasks/Src/remote_control_task.c

chengguizi, 2021-01-04:
```
try parsing newBytes = 18
0 4 20 0 1 98 0 0 0 0 0 0 0 0 0 0 0 4 
try parsing newBytes = 5
0 4 20 0 1 
try parsing newBytes = 13
98 0 0 0 0 0 0 0 0 0 0 0 4 
try parsing newBytes = 18
0 4 20 0 1 98 0 0 0 0 0 0 0 0 0 0 0 4 
try parsing newBytes = 6
0 4 20 0 1 98 
try parsing newBytes = 12
0 0 0 0 0 0 0 0 0 0 0 4 
try parsing newBytes = 18
0 4 20 0 1 98 0 0 0 0 0 0 0 0 0 0 0 4 
try parsing newBytes = 5
0 4 20 0 1 
```

Refer to commit cc6fde892105f9c5abedae46a5edbc396863778e , now we are good to go with parsing the 18 bytes

gimmethisusername, 2021-01-10:
Refer to commit #[https://github.com/nusrobomaster/PX4-Autopilot/commit/cc351bfcb038f20f3429cb2cf2cfe3d1b8046e76](url), 18 bytes from RC successfully parsed and converted to values used in PX4

gimmethisusername, 2021-01-10:
Refer to commit [https://github.com/nusrobomaster/PX4-Autopilot/commit/cc351bfcb038f20f3429cb2cf2cfe3d1b8046e76](url) . Successfully parsed 18 bytes from the remote control and changed the values to fit PX4

chengguizi, 2021-01-10:
@gimmethisusername  I realise this RC input change is only made on Dev A board. Can you test it one the Dev C as well.

@AndreWongZH  Are you able to change the relevant UART settings and test on dev C, before we merge this branch

closed 2021-01-17 by liutiedong