nusrobomaster/PX4-Autopilot #12: CAN Actuator Output

chengguizi, 2021-01-03:
UAVCAN bus seems not to be universal with DJI's CAN Bus protocol, but we could probably reuse some of the CAN library within UAVCAN bus https://github.com/nusrobomaster/PX4-Autopilot/tree/rm2021/src/drivers/uavcan/uavcan_drivers/stm32/driver/include/uavcan_stm32

Could we refer to our last seasons code base for this?

Some code from DJI
https://github.com/RoboMaster/Development-Board-C-Examples/blob/master/14.CAN/application/CAN_receive.c

chengguizi, 2021-01-03:
Some documentation for Motor M3508
https://www.robomaster.com/en-US/products/components/general/M3508

https://rm-static.djicdn.com/tem/17348/RoboMaster%20C620%20Brushless%20DC%20Motor%20Speed%20Controller%20V1.01.pdf

AndreWongZH, 2021-01-07:
https://github.com/nusrobomaster/PX4-Autopilot/commit/eb409919fe4336a6c0764edfe95521fc5a627851
Commit to enable a working can port for dev c

Things done:
* Configure the can port's GPIO pins correctly
* Enable can1 port 
* Set baud rate to 1000000 for can1 port
* calll can_devinit() in the init file

AndreWongZH, 2021-01-07:
To run a can example:
* enable can example in defconfig (application configuration -> example -> can example)

replace the can_main.c (platform/nuttx/Nuttx/apps/examples/can/can_main.c) with the code below:
```
/****************************************************************************
 * examples/can/can_main.c
 *
 *   Copyright (C) 2011 Gregory Nutt. All rights reserved.
 *   Author: Gregory Nutt <gnutt@nuttx.org>
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 *
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in
 *    the documentation and/or other materials provided with the
 *    distribution.
 * 3. Neither the name NuttX nor the names of its contributors may be
 *    used to endorse or promote products derived from this software
 *    without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 * "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 * FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
 * COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 * INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
 * BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS
 * OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED
 * AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
 * ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 *
 ****************************************************************************/

/****************************************************************************
 * Included Files
 ****************************************************************************/

#include <nuttx/config.h>

#include <sys/types.h>
#include <sys/ioctl.h>

#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <string.h>
#include <fcntl.h>
#include <errno.h>
#include <debug.h>

#include <nuttx/can/can.h>

#include "can.h"

/****************************************************************************
 * Pre-processor Definitions
 ****************************************************************************/

#if defined(CONFIG_EXAMPLES_CAN_READ)
#  define CAN_OFLAGS O_RDONLY
#elif defined(CONFIG_EXAMPLES_CAN_WRITE)
#  define CAN_OFLAGS O_WRONLY
#elif defined(CONFIG_EXAMPLES_CAN_READWRITE)
[... 459 lines cut ...]
  return errval;
}

```

After running `can -n 1` output looks like:
```
nmsgs: 1
min ID: 1 max ID: 2047
Bit timing:
      Baud:  1000000
   TSEG1:  6
   TSEG2:  7
       SJW:  1
    ID:     514  DLC:  8
Data0:  20
Data1:  168
Data2:  0
Data3:  0
Data4:  0
Data5:  12
Data6:  23
Data7:  0
Terminating!
```

chengguizi, 2021-01-10:
@AndreWongZH  Can you make the pull request on this?